arhi/EthernetSPI2 #3: Compile Error "cannot declare variable 'server' to be of abstract type 'EthernetServer'"

HilmarSt, 2022-04-19:
While compiling _ESP32EthCamera.ino_ I've got the following error:

<pre>
ESP32EthCamera:81:16: error: cannot declare variable 'server' to be of abstract type 'EthernetServer'
 EthernetServer server(80);
                ^
In file included from D:\ESP32EthCamera\ESP32EthCamera.ino:39:0:
D:\ESP32EthCamera\EthernetSPI2.h:253:7: note:   because the following virtual functions are pure within 'EthernetServer':
 class EthernetServer : public Server {
       ^
In file included from C:\Users\uid\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:152:0,
                 from C:\Users\uid\AppData\Local\Temp\arduino_build_661563\sketch\ESP32EthCamera.ino.cpp:1:
C:\Users\uid\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Server.h:28:18: note: 	virtual void Server::begin(uint16_t)
     virtual void begin(uint16_t port=0) =0;
                  ^

cannot declare variable 'server' to be of abstract type 'EthernetServer'
</pre>

Any idea what's wrong?

arhi, 2022-04-19:
Can not say what's wring without your source. In the sample code included in examples directory line 81 is void setup(){

the general "cannot declare variable 'xxx' to be of abstract type yyy" usually means you are missing some code :) 

Are you sure you have the 
```
#include <SPI.h>
#include <EthernetSPI2.h>
#include "esp_camera.h"
#include "esp_timer.h"
```

sometimes before that line?

is the provided example compiling properly? I just tested here with arduino 1.8.13 ... also I have some old libraries (I do not use arduino env a lot as you can see):
[image]

but before you continue, first try if the example is compiling without any changes, if that does not work than it's something with new libraries

HilmarSt, 2022-04-21:
Sorry for the line number confusion, it's exactly your example and I've only added two comment lines at the beginning - I've corrected my first post.

I'm using the Board "AI Thinker ESP32-CAM" and tried adding your 3 libraries (omitting Esplora) with no success.

**The following changes solve the issue:**
EthernetSPI2.h line 260:
```
	virtual void begin();
->
	virtual void begin(uint16_t port=0);
```

EthernetServer.cpp line 28:
```
void EthernetServer::begin()
->
void EthernetServer::begin(uint16_t port_not_used)
```
...because now they match to:
```
class Server: public Print
{
public:
    virtual void begin(uint16_t port=0) =0;
};
```
in <code>1.0.6\cores\esp32/Server.h</code>

->
_"Sketch uses 371810 bytes (11%) of program storage space."_

I did not yet tested the functionality of the code.

HilmarSt, 2022-04-21:
The ESP32-CAM is working now with Ethernet, thank you for your work!

Two hints:
- I've burnt my skin with the flashlight, I think <code>digitalWrite(FLED, HIGH); delay(1); digitalWrite(FLED, LOW);</code> is long enough for an error indication.
- Programming the ESP32-CAM with the connected W5500 module works if the SCS pin is disconnected during programming.

HilmarSt, 2022-04-21:
I'm using
- Arduino IDE 1.8.18
- Board library "esp32 by Espressif Systems" version 1.0.6

arhi, 2022-04-21:
nice, I'll upgrade local libs and update repo after that :)

closed 2022-04-21 by arhi

pg68390, 2022-10-14:
Bonjour, je viens d'essayer de compiler votre sketch après avoir fait les 2 modifications indiquées ci dessus et je reçois ces messages d'erreur lors de la compilation :+1: 
ESP32EthCamera:79:16: error: cannot declare variable 'server' to be of abstract type 'EthernetServer'
 EthernetServer server(80);
                ^
In file included from /home/philippe/Arduino/EthernetSPI2-master/ESP32EthCamera/ESP32EthCamera.ino:37:0:
/home/philippe/Arduino/libraries/Ethernet/src/EthernetSPI2.h:253:7: note:   because the following virtual functions are pure within 'EthernetServer':
 class EthernetServer : public Server {
       ^
In file included from /home/philippe/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Arduino.h:152:0,
                 from sketch/ESP32EthCamera.ino.cpp:1:
/home/philippe/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Server.h:28:18: note: 	virtual void Server::begin(uint16_t)
     virtual void begin(uint16_t port=0) =0;
                  ^
exit status 1
cannot declare variable 'server' to be of abstract type 'EthernetServer'

pouvez vous m'aider ? cordialement

Philippe