arhi/EthernetSPI2 #4: Mac adress

pg68390, 2022-10-14:
Bonjour, j'aurais besoin d'aide concernant ce sketch notamment concernant cette partie : 

`// Enter a MAC address and IP address for your controller below.
// The IP address will be dependent on your local network:
byte mac[] = {
  0x02, 0xAB, 0xCD, 0xEF, 0x00, 0x01
};
//IPAddress ip(192, 168, 89, 11);
IPAddress ip(192, 168, 89, 11);`

je ne sais quoi mettre dans ces lignes, le MAC adress correspond il à l'adresse mac du module W5500 ou bien celle de ma sortie ethernet du pc ?
pour ip adress peut on mettre une adresse quelconque non déjà utilisée ?

merci pour votre aide et réponses, bien cordialement Philippe

arhi, 2022-10-14:
W5500 does not have MAC address. MAC address is assigned to a network device and is setup in software, since W5500 is just a chip that allows you to make a network device your company need to assign it a MAC address, the start of the MAC address is assigned to your company by the IEEE Registration Authority and then you use rest of the MAC address to assign unique numbers to your devices. MAC address starting with 02 (so 02:00:00:00:00:00 to 02:ff:ff:ff:ff:ff) are reserved for "private" so ideally you should not have any MAC addresses from that range in your LAN except for the devices you made yourself, so make sure you do not use same MAC for 2 devices.

as for the IP, yes, you need to put a static ip not used in your lan that is inside your lan

closed 2022-10-14 by arhi

pg68390, 2022-10-15:
Rebonjour arhi, merci beaucoup pour votre réponse rapide et complète, cela va me permettre de tester votre programme, bonne journée bien cordialement Philippe